saki4510t/UVCCamera #576: Start preview automatically

NhiemHK, 2020-10-24:
How to start preview automatically after first usb camera attached? There is no need to show dialog to select devices because I only need one camera plugged in. Help me, please!

EjazAmir, 2020-11-02:
hey, did you find any solution?

EjazAmir, 2020-11-03:
> How to start preview automatically after first usb camera attached? There is no need to show dialog to select devices because I only need one camera plugged in. Help me, please!

hey, did you find any solution for this?

Guizzz, 2021-05-13:
Yes, i have done it! I write this little function that open the first camera that he find. 
Add it into libusbcamera/com/serenegiant/usb/CameraDialog.java inside the CameraDialog Class.

`public Object getCamera() {
//		mUSBMonitor.dumpDevices();
		updateDevices();
		final List<DeviceFilter> filter = DeviceFilter.getDeviceFilters(getActivity(), R.xml.device_filter);
		mDeviceListAdapter = new DeviceListAdapter(getActivity(), mUSBMonitor.getDeviceList(filter.get(0)));
		Log.v(TAG,mDeviceListAdapter.getItem(0).getDeviceName());
		//mSpinner.setAdapter(mDeviceListAdapter);
		this.dismiss();
		return mDeviceListAdapter.getItem(0);
	}`

And then add these two lines into onCreateDialog function on the same file, before the return.
`               mUSBMonitor.requestPermission((UsbDevice) getCamera());
		((CameraDialogParent)getActivity()).onDialogResult(false);`

It looks like this:
`@Override
    public Dialog onCreateDialog(final Bundle savedInstanceState) {

		final AlertDialog.Builder builder = new AlertDialog.Builder(getActivity());
		builder.setView(initView());
    	builder.setTitle(R.string.select);
	    builder.setPositiveButton(android.R.string.ok, mOnDialogClickListener);
	    builder.setNegativeButton(android.R.string.cancel , mOnDialogClickListener);
	    builder.setNeutralButton(R.string.refresh, null);
	    final Dialog dialog = builder.create();
	    dialog.setCancelable(true);
	    dialog.setCanceledOnTouchOutside(true);

		mUSBMonitor.requestPermission((UsbDevice) getCamera());
		((CameraDialogParent)getActivity()).onDialogResult(false);

        return dialog;
	}`

Of course this is not the best solution but it's work!

chanakamendi, 2022-02-10:
@Guizzz  it doesn't work

Guizzz, 2022-02-11:
I honestly don't understand why the code section in my comment is badly indented and maybe you don't understand.
I updated the comment and i tried to fix it.
I'm sorry but I don't know how to help you more than that.
Also i moved to another solution, this was just a work-around for me.

chanakamendi, 2022-02-11:
@Guizzz  
i really need is when re enter to the activity it should connect camera automatically and show the preview . but when i re enter i get "conntecting" message from   "onConnectDev(UsbDevice device, boolean isConnected)" but i dont get preview . so if i disconnect the camera and connect again it work

chanakamendi, 2022-02-14:
what is your another lib?