saki4510t/UVCCamera #579: How to automatically preview the multiple cameras as they got attached?

EjazAmir, 2020-11-04:
I'm facing an issue, i want to preview the cameras automatically as they got attached to the android device, im using ExampleTest07, Kindly suggest any solution. I'm really stuck at it.

eddykins11, 2020-12-01:
In onAttach(final UsbDevice device), add: mUSBMonitor.requestPermission(device);

EjazAmir, 2020-12-02:
> In onAttach(final UsbDevice device), add: mUSBMonitor.requestPermission(device);

thanks, but what if we have multiple cameras? how to assign cameraview to each usbDevice.

eddykins11, 2020-12-02:
Are you using the same camera model each time or different cameras?

EjazAmir, 2020-12-02:
different camera model each time, if you fixed the issue kindly come in person, as i want to clarify some more things about this, Badly stuck in this. Regards

eddykins11, 2020-12-02:
I doubt this is the most elegant solution but it seems to work; create 2 mOnDeviceConnectListeners and filter each one by the product ID

[MultipleCamActivity.java.zip](https://github.com/saki4510t/UVCCamera/files/5628694/MultipleCamActivity.java.zip)

eddykins11, 2020-12-02:
You can create 2 mOnDeviceConnectListeners and filter each one by productID of your cameras as a quick solution

[MultipleCamActivity.java.zip](https://github.com/saki4510t/UVCCamera/files/5628712/MultipleCamActivity.java.zip)

EjazAmir, 2020-12-02:
thanks, did you also try mulitple cameras in background service?

EjazAmir, 2020-12-02:
if you can help me out in this regard, i will pay you accordingly, but i need help. kindly help me with mulitple cameras in background service too.

eddykins11, 2020-12-02:
I don't think I can help with that and I'd be surprised if that's possible with this library I'm afraid, but maybe someone can correct me.

EjazAmir, 2020-12-02:
yes, check the testExample4

imran855, 2021-01-22:
@eddykins11 

> Are you using the same camera model each time or different cameras?

I am using the same camera model. Is there any suggestion to auto connect?

imran855, 2021-02-08:
I solved my problem. Now it's working fine.

Thanhluc98, 2022-01-11:
> I solved my problem. Now it's working fine.

Hi imran855!
How you can  solve this problem?
I connect the same camera model, but only can open one.