saki4510t/UVCCamera #588: Building with up to date Android SDK and NDK

jumpy88, 2020-12-14:
Hello everyone,
I'm trying to build UVCCamera to use it in a modern Android environment. I cloned Komakai's fork and checked out to komakai-fixes branch. Then I created a local.properties file with SDK and NDK paths
>sdk.dir=/home/<user>/bin/android-sdk/SDK/
ndk.dir=/home/<user>/bin/android-sdk/SDK/ndk/21.3.6528147/

and set `JAVA_HOME=/home/<user>/bin/Java8/jdk1.8.0_271/`. When I run `gradle build` it fails reporting
>Ran lint on variant release: 13 issues found
Ran lint on variant debug: 13 issues found
Wrote HTML report to file:///home/<user>/bin/UVCCamera/UVCCamera_komakai/libuvccamera/build/reports/lint-results.html
Wrote XML report to file:///home/<user>/bin/UVCCamera/UVCCamera_komakai/libuvccamera/build/reports/lint-results.xml
:libuvccamera:lint FAILED
FAILURE: Build failed with an exception.
\* What went wrong:
Execution failed for task ':libuvccamera:lint'.
\> Lint found errors in the project; aborting build.
  Fix the issues identified by lint, or add the following to your build script to proceed with errors:
  ...
  android {
      lintOptions {
          abortOnError false
      }
  }
  ...
  The first 3 errors (out of 5) were:
  /home/<user>/bin/UVCCamera/UVCCamera_komakai/libuvccamera/src/main/java/com/serenegiant/usb/USBMonitor.java:891: Error: Call requires API level 21 (current min is 14): android.hardware.usb.UsbDevice#getManufacturerName [NewApi]
      info.manufacturer = device.getManufacturerName();
                                 ~~~~~~~~~~~~~~~~~~~
  /home/<user>/bin/UVCCamera/UVCCamera_komakai/libuvccamera/src/main/java/com/serenegiant/usb/USBMonitor.java:892: Error: Call requires API level 21 (current min is 14): android.hardware.usb.UsbDevice#getProductName [NewApi]
      info.product = device.getProductName();
                            ~~~~~~~~~~~~~~
  /home/<user>/bin/UVCCamera/UVCCamera_komakai/libuvccamera/src/main/java/com/serenegiant/usb/USBMonitor.java:893: Error: Call requires API level 21 (current min is 14): android.hardware.usb.UsbDevice#getSerialNumber [NewApi]
      info.serial = device.getSerialNumber();
                           ~~~~~~~~~~~~~~~

There is also a warning while building
>Android NDK: WARNING: APP_PLATFORM android-16 is higher than android:minSdkVersion 14 in ./AndroidManifest.xml. NDK binaries will *not* be compatible with devices older than android-16. See https://android.googlesource.com/platform/ndk/+/master/docs/user/common_problems.md for more information.

I tried to modify `minSdkVersion` value but other errors rose, so I decided to rollback. Any suggestion how to get rid of this?

komakai, 2020-12-14:
@jumpy88 please check out 5f792118e6413f2c7f5c0547844844fb4806cb08

jumpy88, 2020-12-15:
It's the commit where I'm. Maybe I'm configuring something the wrong way. Perhaps `ndk.dir=/home//bin/android-sdk/SDK/ndk/21.3.6528147/` is too new/old. Which API level are you using?

komakai, 2020-12-15:
@jumpy88 I don't think the problem is with the NDK version - please try building with Android Studio and see if that helps. Note that the errors you are seeing are from the linter so they probably OK to ignore

jumpy88, 2020-12-15:
You are right, Android Studio builds it (I don't understand why `gradle build` doesn't but at moment this doesn't matter). I tried to install all of the examples: usbCameraTest0 - 8 immediately crash at startup, usbCameraTest (with no number) starts and seems to work, but not for all of my USB cameras. A 640x480 camera works fine (is recognized and show pictures) but the 2560x960, I need to work with, doesn't. I couldn't find any useful error in logcat, but I suspect the sample code doesn't support the strange or too high resolution. Any suggestion is welcome.

jumpy88, 2020-12-15:
Maybe I found something but I don't what to do: `E/OculusCameraDriver: mm_stream_get_v4l2_fmt: Unknown fmt=112`

jumpy88, 2020-12-16:
Perhaps I've also found why the others don't start:
>2020-12-16 02:59:18.419 15178-15178/com.serenegiant.usbcameratest2 E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.serenegiant.usbcameratest2, PID: 15178
    java.lang.RuntimeException: Unable to start activity ComponentInfo{com.serenegiant.usbcameratest2/com.serenegiant.usbcameratest2.MainActivity}: java.lang.IllegalStateException: You need to use a Theme.AppCompat theme (or descendant) with this activity.
        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2680)
        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2741)
        at android.app.ActivityThread.-wrap12(ActivityThread.java)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1478)
        at android.os.Handler.dispatchMessage(Handler.java:102)
        at android.os.Looper.loop(Looper.java:154)
        at android.app.ActivityThread.main(ActivityThread.java:6165)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:886)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:776)
     Caused by: java.lang.IllegalStateException: You need to use a Theme.AppCompat theme (or descendant) with this activity.
        at android.support.v7.app.AppCompatDelegateImplV9.createSubDecor(AppCompatDelegateImplV9.java:354)
        at android.support.v7.app.AppCompatDelegateImplV9.ensureSubDecor(AppCompatDelegateImplV9.java:323)
        at android.support.v7.app.AppCompatDelegateImplV9.setContentView(AppCompatDelegateImplV9.java:284)
        at android.support.v7.app.AppCompatActivity.setContentView(AppCompatActivity.java:139)
        at com.serenegiant.usbcameratest2.MainActivity.onCreate(MainActivity.java:85)
        at android.app.Activity.performCreate(Activity.java:6724)
        at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1119)
        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2633)
        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2741) 
        at android.app.ActivityThread.-wrap12(ActivityThread.java) 
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1478) 
        at android.os.Handler.dispatchMessage(Handler.java:102) 
        at android.os.Looper.loop(Looper.java:154) 
        at android.app.ActivityThread.main(ActivityThread.java:6165) 
        at java.lang.reflect.Method.invoke(Native Method) 
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:886) 
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:776)

komakai, 2020-12-16:
> Maybe I found something but I don't what to do: E/OculusCameraDriver: mm_stream_get_v4l2_fmt: Unknown fmt=112

there is no file with the name `mm_stream_get_v4l2_fmt` in this project so that message comes from somewhere else

komakai, 2020-12-16:
> Perhaps I've also found why the others don't start:
>> Caused by: java.lang.IllegalStateException: You need to use a Theme.AppCompat theme (or descendant) with this activity.

OK - so copy the relevant changes from the usbcameratest1 - should be easy enough

jumpy88, 2020-12-16:
Ok, this fixed almost all of them; usbCameraTest2 also required to remove `android:background="#ff000000"` in `activity_main.xml > com.serenegiant.widget.SimpleUVCCameraTextureView` (line 39) because background seems not to be supported by `TextureView`. Some samples still don't show any image and sometimes randomly crash but I don't want to debug all of them. I prefer to move on and try to understand why no one works with my other USB camera. I go to investigate about aspect ratio that I think could be the problem. @komakai do you know if there is any constraint about it in UVCCamera?

jumpy88, 2020-12-19:
Fixed it: it was sufficient to modify `DEFAULT_PREVIEW_WIDTH` and `DEFAULT_PREVIEW_HEIGHT` in  `libuvccamera/src/main/java/com/serenegiant/usb/UVCCamera.java`