saki4510t/UVCCamera #590: Fatal exception loading libjpeg-turbo, libusb and libuvc

emejoti, 2020-12-17:
Hi!
I'm trying the usbCameraTest0 on my Android and when I select the USB camera it gave me the next error:

`E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.serenegiant.usbcameratest0, PID: 13261
    java.lang.UnsatisfiedLinkError: dalvik.system.PathClassLoader[DexPathList[[zip file "/system/framework/org.apache.http.legacy.boot.jar", zip file "/data/app/com.serenegiant.usbcameratest0-AeCN71iEUUMdhjpfB3peuA==/base.apk"],nativeLibraryDirectories=[/data/app/com.serenegiant.usbcameratest0-AeCN71iEUUMdhjpfB3peuA==/lib/arm64, /system/lib64]]] couldn't find "libjpeg-turbo1500.so"
        at java.lang.Runtime.loadLibrary0(Runtime.java:1012)
        at java.lang.System.loadLibrary(System.java:1669)
        at com.serenegiant.usb.UVCCamera.<clinit>(UVCCamera.java:121)
        at com.serenegiant.usbcameratest0.MainActivity$2$1.run(MainActivity.java:182)
        at android.os.Handler.handleCallback(Handler.java:873)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:193)
        at android.app.ActivityThread.main(ActivityThread.java:6746)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:858)`

It happened in System.loadLibrary("jpeg-turbo1500"); of UVCCamera class.

I'll appreciate any help, thanks.

jumpy88, 2020-12-30:
Hi @emejoti, check if these libraries are included in your apk or not. You can browse it with an archive manager.

jumpy88, 2020-12-30:
Also, remember to check it is there for the right architecture (arm64-v8a, armeabi-v7a...) matching your set up.

emejoti, 2021-01-05:
Thanks for your comments, the error was on the gradle.

closed 2021-01-05 by emejoti

jumpy88, 2021-01-05:
Ok. How did you solve it? Just in case someone else met the same error.

emejoti, 2021-01-05:
I guess I did something wrong  when I include the libraries, I fixed it changing the source in the libuvccamera build gradle.

zipswich, 2022-01-15:
In my case, I have to add arm64-v8a to Application.mk under UVCCamera\libuvccamera\src\main\jni
`APP_ABI := armeabi-v7a arm64-v8a x86`

mrmuke, 2022-06-25:
@emejoti I'm having the same problem, can I ask what you changed the source to when you included the libraries? Thank you so much - I've been struggling with this one for a little while.