saki4510t/UVCCamera #595: Getting frames with IFrameCallback

emejoti, 2021-01-05:
Hello, 

I need to get the frames from the camera as a ByteBuffer, so I'm using IFrameCallback like this:

```
mUVCCamera.setFrameCallback(mIFrameCallback1, UVCCamera.PIXEL_FORMAT_YUV420SP);

private final IFrameCallback mIFrameCallback1 = new IFrameCallback() {
			@Override
			public void onFrame(final ByteBuffer frame) {
                             // Saving the frame on a ConcurrentLinkedDeque 
			}
		};
```

The thing is, when I get the frame from that cocurrentLinkedDeque, because I need to use it on another function, the frame looks wrong. I tried all the format on UVCCamera and YUV420SP is the one that gave me less problems.

Here are some of the frames I get:

[image]
[image]
[image]

I'll appreciate any help, thanks.

emejoti, 2021-01-07:
Solved! Here is the code:
```
     private int buf_sz = 4096;
     private byte[] buf = new byte[buf_sz];

    private final IFrameCallback mIFrameCallback1 = new IFrameCallback() {
        @Override
        public void onFrame(final ByteBuffer frame) {
            synchronized (mSync) {
                final int n = frame.limit();
                if (buf_sz < n) {
                    buf_sz = n;
                    buf = new byte[n];
                }
                frame.get(buf, 0, n);

                //Save buf on my ConcurrentLinkedDeque 
            }
        }
    };
```

And then just transform the byte array into ByteBuffer with `ByteBuffer.wrap(buf );`

I found the solution here: https://github.com/saki4510t/UVCCamera/issues/241#issuecomment-336334886

closed 2021-01-07 by emejoti

rgm-22, 2021-10-02:
Hi @emejoti!

How did you succeed with this on the MainActivity?

"And then just transform the byte array into ByteBuffer with ByteBuffer.wrap(buf );"

I would like to record what I am displaying on IFrameCallback.

Thank you very much in advance!