saki4510t/UVCCamera #603: Crash after running for a long time

HolmesJJ, 2021-03-31:
When I use a USB external camera to test, after a long time of running, the app sometimes crashes. Has this problem occurred before?

bandrarom, 2021-12-08:
have you fix it?

HolmesJJ, 2021-12-09:
> have you fix it?

You can try this link: https://github.com/jiangdongguo/AndroidUSBCamera

mking123456, 2022-03-31:
https://github.com/jiangdongguo/AndroidUSBCamera, is it work for you? i tried this project demo on rk3288 board a little while，the system crashed and restart.

HolmesJJ, 2022-03-31:
> 

I tried it on my Android Phone, it works fine.

HolmesJJ, 2022-03-31:
> https://github.com/jiangdongguo/AndroidUSBCamera, is it work for you? i tried this project demo on rk3288 board a little while，the system crashed and restart.

I tried it on my Android Phone, it works fine.