saki4510t/UVCCamera #605: How to solve the preview delay problem

along1991, 2021-04-13:
The 1080p 30fps mjpeg camera preview time delay is 400ms,How to solve it?

along1991, 2021-04-19:
> 1080p 30fps mjpeg摄像机预览时间延迟为400ms，如何解决？

closed 2021-04-19 by along1991

zyowo, 2021-05-07:
我也有延迟相关的问题，请问您后来解决了吗？

along1991, 2021-05-24:
> 我也有延迟相关的问题，请问您后来解决了吗？

解决了,问题出再mjpeg转码yuyv再转码rgb这里耗时很久,,我这里使用libyuv库,mjpeg转码i420再转码abgr,延时降低到了130ms

YZ-Inspify, 2021-10-05:
@along1991 - 请问有示例代码如何把获得的`uvc_frame`在libyuv中mjpeg转i420再转abgr吗？

多谢帮助！

along1991, 2021-10-14:
> @along1991 - 有示例代码如何把获得的`uvc_frame`在libyuv中mjpeg转i420再转abgr吗？
> 
> 多谢帮助！
参考这篇文章https://blog.csdn.net/xuhui_7810/article/details/103741722?
里面搜MJPGToI420就行

liubilly2021, 2022-11-26:
大家好，android uvccamera MJPEG 2592x1944 预览延迟有400ms ，分辨率越高，延迟约到，请大家帮忙看看，怎么优化。

liubilly2021, 2022-11-28:
> > 我也有延迟相关的问题，请问您后来解决了吗？
> 
> 解决了,问题出再mjpeg转码yuyv再转码rgb这里耗时很久,,我这里使用libyuv库,mjpeg转码i420再转码abgr,延时降低到了130ms

你好，这个有参考到吗放出来吗

along1991, 2022-11-28:
你用usb测一下延时，就是沈垚的那个https://app.xiaomi.com/details?id=com.shenyaocn.android.usbcamera，看下延时是多少，如果差不多就不用改了，如果差的比较多，那就改改

---Original---
From: ***@***.***&gt;
Date: Mon, Nov 28, 2022 16:53 PM
To: ***@***.***&gt;;
Cc: ***@***.******@***.***&gt;;
Subject: Re: [saki4510t/UVCCamera] How to solve the preview delay problem(#605)

zhouqunhuang, 2024-05-10:
你好请问你能帮我解决一下吗？我在4K上延迟有一两秒。如果解决成功给你报酬。