saki4510t/UVCCamera #628: target version >=30 not work

heaveninfosoft, 2021-12-06:
hello for new app or update app must be target version >=30 .
but when i set this in app it always crash

dinhtung249, 2021-12-11:
I reached sdk 30 and it is ok. Show  your logcat

heaveninfosoft, 2021-12-11:
@dinhtung249 
java.lang.SecurityException: User has not given 10146/ permission to access device /dev/bus/usb/002/003
        at android.os.Parcel.createException(Parcel.java:2071)
        at android.os.Parcel.readException(Parcel.java:2039)
        at android.os.Parcel.readException(Parcel.java:1987)
        at android.hardware.usb.IUsbSerialReader$Stub$Proxy.getSerial(IUsbSerialReader.java:123)
        at android.hardware.usb.UsbDevice.getSerialNumber(UsbDevice.java:143)
        at com.serenegiant.usb.USBMonitor.getDeviceKeyName(USBMonitor.java:733)
        at com.serenegiant.usb.USBMonitor.getDeviceKey(USBMonitor.java:766)
        at com.serenegiant.usb.USBMonitor.updatePermission(USBMonitor.java:469)
        at com.serenegiant.usb.USBMonitor.hasPermission(USBMonitor.java:459)
        at com.serenegiant.usb.USBMonitor$2.run(USBMonitor.java:596)
        at android.os.Handler.handleCallback(Handler.java:883)
        at android.os.Handler.dispatchMessage(Handler.java:100)
        at android.os.Looper.loop(Looper.java:214)
        at android.os.HandlerThread.run(HandlerThread.java:67)

it denied permission

Ivan421399, 2021-12-23:
in _libuvccamera/src/java/com/serenegiant/usb/USBMonitor_ 
on row 662

comment out this code
```
if (TextUtils.isEmpty(serial)) {
   sb.append(device.getSerialNumber());	sb.append("#");	// API >= 21
}
```

Hope it helps

heaveninfosoft, 2021-12-23:
@Ivan421399 bro have you tested after comment out that line??
i already did that but then it  crashed problem solved but not show preview

Ivan421399, 2021-12-23:
@heaveninfosoft When I used uvc cam with a usb mic (through a hub) the lib was crashing in the same way. Maybe you have different issue.

heaveninfosoft, 2021-12-23:
@Ivan421399 no no my straight forward question is when target above = 30 lib crash at getSerialNumber().
now when i comment out as you suggested then not crash that ok , but then nothing happen i mean no preview

heaveninfosoft, 2021-12-23:
@Ivan421399 please bro help me

Ivan421399, 2021-12-23:
@heaveninfosoft You know. I was testing my app on various devices with android 11 and I have no image either.
My crashes were happening on devices with android 10 too but it was because of the mic, though as you say crashes are gone now.
I was thinking it's the librarys issue that one cannot get image on later android versions.
What is the resolution of your usb cam that you tested?

heaveninfosoft, 2021-12-23:
640P  and 720P   there no resolution issue i confirmed.
but let me know if you got preview on android 10 or 11 when target >=29

Ivan421399, 2021-12-23:
@heaveninfosoft welp, good to know that such a small res video is still not going through, because our cam is fullhd mjpeg with huge fat buffers and we thought it's the channel capacity of the usb issue.
Sadly I don't know how to help you, moreover I don't know how to fix our problem either. I'm still searching for the answer.

Ivan421399, 2021-12-23:
btw on top of that using this library can provoke the phone to reboot.
It only happens on new android version. On old Redmi 9 with android 11 the issue persists but on older Poko M3 with android 10 everything is fine.

heaveninfosoft, 2021-12-23:
i have poco F1 android 10 when i set target=>29 it not work . here lib author never reply.
i am surprise that now day android 10,11 common and for play store publish target =>29 mandatory still no one raise this issue

Ivan421399, 2021-12-23:
@heaveninfosoft I can't say that it will help you but at least it's something.
https://github.com/alejandrorosas/android-uvc-rtmp-stream/tree/master/libraries/libuvccamera
This repo uses this library but with various updates. Those updates are pretty recent so give it a try.

Ivan421399, 2022-01-10:
> I reached sdk 30 and it is ok.

Can you please tell me on which phone you launched the lib?

heaveninfosoft, 2022-01-10:
@Ivan421399 bro let me know weather it work or not above sdk  29 or 30  because main point here is when we target 30 then it crashed

Ivan421399, 2022-01-17:
@heaveninfosoft According to our tests this lib doesn't work on devices with MediaTek CPUs alongside with android 11.
For example the Samsung A12 (125-f) with MediaTek cpu displays green screen, but the different revision of this phone: 127-f (which is also called Nacho) displays everything fine. Both of those phones had Android 11.
We also tested several other phones and every single one which has MediaTek cpu displayed green screen.

heaveninfosoft, 2022-01-18:
@Ivan421399 bro issue with usb camera permission related after target 28 . it crash on USB camera Permission dialog

ToysoftInc, 2022-10-30:
I got the same problem. Crashes on >=30 api.
I tracked down the problem.  It has to do with the Intent when calling

usbMonitor.register()

in USBMonitor class

`				mPermissionIntent = PendingIntent.getBroadcast(context, 0, new Intent(ACTION_USB_PERMISSION), 0);
`
the Flag can not be 0

I'm not able to compile the new lib

YAsheep, 2024-11-01:
> in _libuvccamera/src/java/com/serenegiant/usb/USBMonitor_ on row 662
> 
> comment out this code
> 
> ```
> if (TextUtils.isEmpty(serial)) {
>    sb.append(device.getSerialNumber());	sb.append("#");	// API >= 21
> }
> ```
> 
> Hope it helps

Thanks, it's real help.  do you know how to handle this crash?:
  java.lang.UnsupportedOperationException: open failed:result=-1