saki4510t/UVCCamera #629: any one's saki450t/uvcamera  work properly?

bandrarom, 2021-12-09:


jeranio, 2021-12-13:
Yes, mine does.
Using these settings :

	supportLibVersion = '27.1.1'  // variable that can be referenced to keep support libs consistent
	commonLibVersion= '4.1.1'
	versionBuildTool = '27.0.3'
	versionCompiler = 31
	versionTarget = 27
	versionNameString = '1.0.0'
	javaSourceCompatibility = JavaVersion.VERSION_1_8
	javaTargetCompatibility = JavaVersion.VERSION_1_8
------------------------------------------------------------------------------------
allprojects {
    repositories {
//    	google()
		maven { url 'https://maven.google.com' }
		jcenter()
		maven { url 'https://raw.github.com/saki4510t/libcommon/master/repository/' }
    }
}

--------------------------------------------------------------------------------------------------------------

/Android/Sdk/ndk-bundle/android-ndk-r20b

---------------------------------------------------------------------------------------------------------------

Android Studio Arctic Fox 2020.3.1 Patch 2

bandrarom, 2021-12-13:
thank you for you comment

jeranio, 2021-12-14:
Also in manifest files change theme to Theme.Appcompat and remove background colors from layout files.

bandrarom, 2021-12-14:
yeah thank you for your reply . can i know how to devide audio from     mCameraHelper.setOnPreviewFrameListener(new AbstractUVCCameraHandler.OnPreViewResultListener() {
                @Override
                public void onPreviewResult(byte[] bytes) {}

i want to get uvc camera audio seperatly

lifei, 2022-03-08:
> Yes, mine does. Using these settings :
> 
> ```
> supportLibVersion = '27.1.1'  // variable that can be referenced to keep support libs consistent
> commonLibVersion= '4.1.1'
> versionBuildTool = '27.0.3'
> versionCompiler = 31
> versionTarget = 27
> versionNameString = '1.0.0'
> javaSourceCompatibility = JavaVersion.VERSION_1_8
> javaTargetCompatibility = JavaVersion.VERSION_1_8
> ```
> 
> allprojects { repositories { // google() maven { url 'https://maven.google.com' } jcenter() maven { url 'https://raw.github.com/saki4510t/libcommon/master/repository/' } } }
> 
> /Android/Sdk/ndk-bundle/android-ndk-r20b
> 
> Android Studio Arctic Fox 2020.3.1 Patch 2 classpath 'com.android.tools.build:gradle:7.0.2' distributionUrl=https://services.gradle.org/distributions/gradle-7.0.2-all.zip

save my time. thank you.