saki4510t/UVCCamera #631: can i get audio data from   mCameraHelper.setOnPreviewFrameListener()?

bandrarom, 2021-12-22:


jiaozebo, 2022-01-11:
use AudioRecord to read audio

Qamar4P, 2023-10-25:
@saki4510t 请给我们一段代码片段或想法，说明如何在 Playstore 上的 USB 相机应用程序中从 UVC 相机实时播放音频
https://play.google.com/store/apps/details?id=com.shenyaocn.android.usbcamera&hl=en&gl=US

Clark-caipeiyuan, 2023-10-25:
收到了，谢谢！

Qamar4P, 2023-10-25:
如果音频播放逻辑是付费的，那么我们将购买这个库以在我们的商业产品中使用...请联系 qamar4p@gmail.com

Qamar4P, 2023-10-26:
@Clark-caipeiyuan @saki4510t @dragonman225 @Enny1991 @WangXuan95 @rehlma @w-shackleton @wf9a5m75 community guys & USB & UVC protocol experts please guide & share the solution to UVC audio interface live playback same as USB Camera app... We are using USB Capture device see screenshots. Thanks & Regards

[image]
[image]