saki4510t/UVCCamera #633: android.graphics.SurfaceTexture com.serenegiant.usb.widget.CameraViewInterface.getSurfaceTexture()' on a null object reference

bandrarom, 2022-01-05:
E/AndroidRuntime: FATAL EXCEPTION: main
    Process: io.agora.tutorials1v1vcall, PID: 18010
    java.lang.NullPointerException: Attempt to invoke interface method 'android.graphics.SurfaceTexture com.serenegiant.usb.widget.CameraViewInterface.getSurfaceTexture()' on a null object reference
        at com.jiangdg.usbcamera.UVCCameraHelper.startPreview(UVCCameraHelper.java:324)
        at io.agora.tutorials1v1vcall.ui.slideshow.VideoCamerTesterFragment.onCreateView(VideoCamerTesterFragment.java:124)
        at androidx.fragment.app.Fragment.performCreateView(Fragment.java:2698)
        at androidx.fragment.app.FragmentStateManager.createView(FragmentStateManager.java:320)
        at androidx.fragment.app.FragmentManager.moveToState(FragmentManager.java:1187)
        at androidx.fragment.app.FragmentManager.addAddedFragments(FragmentManager.java:2224)
        at androidx.fragment.app.FragmentManager.executeOpsTogether(FragmentManager.java:1997)
        at androidx.fragment.app.FragmentManager.removeRedundantOperationsAndExecute(FragmentManager.java:1953)
        at androidx.fragment.app.FragmentManager.execPendingActions(FragmentManager.java:1849)
        at androidx.fragment.app.FragmentManager$4.run(FragmentManager.java:413)
        at android.os.Handler.handleCallback(Handler.java:938)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:246)
        at android.app.ActivityThread.main(ActivityThread.java:8595)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:602)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1130)

i tried to make this app on fragment and i get this error

jiaozebo, 2022-01-11:
a SurfaceView may not be necessary 
you may try this repo:
https://github.com/tsinglink/uvccamera
and test with this apk:
https://github.com/tsinglink/uvccamera/releases/download/1.0/simpleuvccamera-release.apk

KanwalKhan-20, 2022-01-12:
> a SurfaceView may not be necessary you may try this repo: https://github.com/tsinglink/uvccamera and test with this apk: https://github.com/tsinglink/uvccamera/releases/download/1.0/simpleuvccamera-release.apk

It is not working :(

jiaozebo, 2022-01-12:
maybe  the compatibility of the uvc camera is not very good. or maybe  the compatibility of all USB-related devices is not good. it's very depressing, sometimes you have to dig into the very deep level of c code.

jiaozebo, 2022-01-12:
@KanwalKhan-20 I see a device or a camera,  can't remember, Printing something like "mjpeg data is not legal". I suspect that the usb protocol is unreliable and will lose data. but I'm not sure