saki4510t/UVCCamera #634: Android 11 and 12

KanwalKhan-20, 2022-01-10:
No Preview after inserting the camera in Android 11 and 12.
What should be the compile and target sdk to get the preview ?

jiaozebo, 2022-01-11:
you may try this repo:
https://github.com/tsinglink/uvccamera
and test with this apk:
https://github.com/tsinglink/uvccamera/releases/download/1.0/simpleuvccamera-release.apk

jiaozebo, 2022-01-11:
compileSdkVersion 28

    defaultConfig {
        applicationId "com.example.simpleuvccamera"
        minSdkVersion 21
        targetSdkVersion 28
        versionCode 1
        versionName "1.0"

        signingConfig signingConfigs.debug
        testInstrumentationRunner "androidx.test.runner.AndroidJUnitRunner"
    }

KanwalKhan-20, 2022-01-11:
> you may try this repo: https://github.com/tsinglink/uvccamera and test with this apk: https://github.com/tsinglink/uvccamera/releases/download/1.0/simpleuvccamera-release.apk

Same preview is not available

Ivan421399, 2022-01-17:
@KanwalKhan-20
Can you tell the cpu's of the phones you were using?
It seems like the MediaTek processors have this issue with android 11, 12.

KanwalKhan-20, 2022-01-17:
> @KanwalKhan-20 Can you tell the cpu's of the phones you were using? It seems like the MediaTek processors have this issue with android 11, 12.

**Snapdragon 855**
Model Name : Galaxy S10

Ivan421399, 2022-01-17:
@KanwalKhan-20 
have you tried to change the format, maybe the webcam is MJPEG?

KanwalKhan-20, 2022-01-17:
> @KanwalKhan-20 have you tried to change the format, maybe the webcam is MJPEG? Or you already had it working on other phones?

@Ivan421399 how to change the format ? Are you asking me to change the camera ?

KanwalKhan-20, 2022-01-17:
@Ivan421399 attached are the specifications of camera which I am using.
[image]

Ivan421399, 2022-01-18:
Well I don't know exactly which example or whatever you're using.
I don't remember exactly regarding the examples but if there are settings that you can change from the app interface try to change them.
In code however, find the UVCCamera object. Then in it's method "setPreviewSize" try to change the third argument to either FRAME_FORMAT_MJPEG or FRAME_FORMAT_YUYV. Also try to change the resolution in the same method.

Sumeyya-Eser, 2022-08-03:
> Well I don't know exactly which example or whatever you're using.
> I don't remember exactly regarding the examples but if there are settings that you can change from the app interface try to change [...]

Do you have any update ?

lichao3140, 2023-02-01:
> you may try this repo: https://github.com/tsinglink/uvccamera and test with this apk: https://github.com/tsinglink/uvccamera/releases/download/1.0/simpleuvccamera-release.apk

web page is 404

BekhruzDev, 2023-06-22:
@ghost Have you solved this problem? I am having this issue too, still not previewing on Android 12 and 13 devices

Utsav-Mehta, 2024-04-06:
@BekhruzDev Have you ended up finding any solution?