saki4510t/UVCCamera #635: monitor.mUsbManager.openDevice(device) returns null

zipswich, 2022-01-18:
```
		/**
		 * this class needs permission to access USB device before constructing
		 * @param monitor
		 * @param device
		 */
		private UsbControlBlock(final USBMonitor monitor, final UsbDevice device) {
			if (DEBUG) Log.i(TAG, "UsbControlBlock:constructor");
			mWeakMonitor = new WeakReference<USBMonitor>(monitor);
			mWeakDevice = new WeakReference<UsbDevice>(device);
			mConnection = monitor.mUsbManager.openDevice(device);
			mInfo = updateDeviceInfo(monitor.mUsbManager, device, null);
```
monitor.mUsbManager.openDevice(device) in the above code of USBMonitor.java returns null. The most relevant logcat part:
```
2022-01-17 16:54:53.078 1476-1731/? D/OpenGLRenderer: endAllActiveAnimators on 0x7434380000 (RippleDrawable) with handle 0x7432178280
2022-01-17 16:54:53.129 972-1252/? I/CameraProviderManager: Camera device device@3.4/external//dev/video5 status is now NOT_PRESENT, was PRESENT
2022-01-17 16:54:53.129 972-1252/? I/CameraService: onDeviceStatusChanged: Status changed for cameraId=/dev/video5, newStatus=0
2022-01-17 16:54:53.130 972-1252/? W/CameraService: removeClientLocked: Could not evict client, no client for camera ID /dev/video5
2022-01-17 16:54:53.130 972-1252/? W/CameraFlashlight: hasFlashUnitLocked: camera /dev/video5 not present when findFlashUnits() was called
2022-01-17 16:54:53.143 3366-3397/? E/USBMonitor: could not connect to device /dev/bus/usb/001/003
2022-01-17 16:54:53.143 3366-3397/? V/MainActivity: onConnect:
2022-01-17 16:54:53.144 3366-3400/? V/CameraThread: handleOpen:
2022-01-17 16:54:53.144 3366-3400/? V/CameraThread: handleClose:
2022-01-17 16:54:53.144 3366-3400/? V/CameraThread: handleStopRecording:mMuxer=null
2022-01-17 16:54:53.144 3366-3400/? V/UVCCameraTextureView: setVideoEncoder:
2022-01-17 16:54:53.144 3366-3425/? V/UVCCameraTextureView: RenderThread#setEncoder:encoder=null
2022-01-17 16:54:53.145 1296-1465/? D/UsbHostManager: Removed device at /dev/bus/usb/001/003: Boson
2022-01-17 16:54:53.145 1296-1465/? I/UsbAlsaManager: USB Audio Device Removed: null
```
The following from logcat provides more background:
```
2022-01-17 16:54:42.788 650-737/? E/ExtCamDev@3.4: getFrameRateList: failed to get supported frame rates for format:YU12 w 640 h 512
2022-01-17 16:54:42.789 650-737/? E/ExtCamDev@3.4: trimSupportedFormats: input format list is empty!
2022-01-17 16:54:42.790 650-737/? I/CamPvdr@2.4-external: ExtCam: adding /dev/video5 to External Camera HAL!
2022-01-17 16:54:42.792 972-1252/? I/CameraProviderManager: Enumerating new camera device: device@3.4/external//dev/video5
2022-01-17 16:54:42.793 650-737/? E/ExtCamDev@3.4: getFrameRateList: failed to get supported frame rates for format:YU12 w 640 h 512
2022-01-17 16:54:42.796 650-737/? E/ExtCamDev@3.4: trimSupportedFormats: input format list is empty!
2022-01-17 16:54:42.798 1296-1465/? I/UsbDescriptorParser: Unknown Descriptor len: 13 type:0x24
2022-01-17 16:54:42.798 972-1252/? I/CameraService: onDeviceStatusChanged: Status changed for cameraId=/dev/video5, newStatus=1
2022-01-17 16:54:42.798 1296-1465/? I/UsbDescriptorParser: Unknown Descriptor len: 17 type:0x24
2022-01-17 16:54:42.798 972-1252/? I/CameraService: onDeviceStatusChanged: Unknown camera ID /dev/video5, a new camera is added
2022-01-17 16:54:42.798 1296-1465/? I/UsbDescriptorParser: Unknown Descriptor len: 9 type:0x24
2022-01-17 16:54:42.798 1296-1465/? I/UsbDescriptorParser: Unknown Descriptor len: 11 type:0x24
2022-01-17 16:54:42.798 972-1252/? W/CameraFlashlight: hasFlashUnitLocked: camera /dev/video5 not present when findFlashUnits() was called
2022-01-17 16:54:42.798 1296-1465/? I/UsbDescriptorParser: Unknown Descriptor len: 17 type:0x24
2022-01-17 16:54:42.798 1296-1465/? I/UsbDescriptorParser: Unknown Descriptor len: 27 type:0x24
2022-01-17 16:54:42.798 1296-1465/? I/UsbDescriptorParser: Unknown Descriptor len: 34 type:0x24
2022-01-17 16:54:42.798 1296-1465/? I/UsbDescriptorParser: Unknown Descriptor len: 27 type:0x24
2022-01-17 16:54:42.798 1296-1465/? I/UsbDescriptorParser: Unknown Descriptor len: 34 type:0x24
2022-01-17 16:54:42.798 1296-1465/? I/UsbDescriptorParser: Unknown Descriptor len: 27 type:0x24
2022-01-17 16:54:42.798 1296-1465/? I/UsbDescriptorParser: Unknown Descriptor len: 34 type:0x24
2022-01-17 16:54:42.798 1296-1465/? I/UsbDescriptorParser: Unknown Descriptor len: 27 type:0x24
2022-01-17 16:54:42.798 1296-1465/? I/UsbDescriptorParser: Unknown Descriptor len: 34 type:0x24
2022-01-17 16:54:42.798 1296-1465/? I/UsbDescriptorParser: Unknown Descriptor len: 6 type:0x24
2022-01-17 16:54:42.798 1296-1465/? I/UsbDescriptorParser: Unknown Descriptor len: 5 type:0x24
2022-01-17 16:54:42.798 1296-1465/? I/UsbDescriptorParser: Unknown Descriptor len: 4 type:0x24
2022-01-17 16:54:42.798 1296-1465/? I/UsbDescriptorParser: Unknown Descriptor len: 5 type:0x24
2022-01-17 16:54:42.802 1296-1465/? D/UsbHostManager: USB device attached: vidpid 09cb:4007 mfg/product/ver/serial FLIR/Boson/1.00/58283 hasAudio/HID/Storage: false/false/false
2022-01-17 16:54:42.798 1296-1465/? I/UsbDescriptorParser: Unknown Descriptor len: 5 type:0x24
2022-01-17 16:54:42.804 1296-1465/? D/UsbDeviceDescriptor:   1 configs
2022-01-17 16:54:42.809 1296-1465/? D/UsbHostManager: Added device UsbDevice[mName=/dev/bus/usb/001/003,mVendorId=2507,mProductId=16391,mClass=239,mSubclass=2,mProtocol=1,mManufacturerName=FLIR,mProductName=Boson,mVersion=1.00,mSerialNumber=58283,mConfigurations=[
    UsbConfiguration[mId=1,mName=null,mAttributes=192,mMaxPower=250,mInterfaces=[
    UsbInterface[mId=0,mAlternateSetting=0,mName=FLIR Camera,mClass=14,mSubclass=1,mProtocol=0,mEndpoints=[]
    UsbInterface[mId=1,mAlternateSetting=0,mName=Video In (disabled),mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=129,mAttributes=2,mMaxPacketSize=512,mInterval=0]]
    UsbInterface[mId=2,mAlternateSetting=0,mName=Serial port interface,mClass=2,mSubclass=2,mProtocol=1,mEndpoints=[
    UsbEndpoint[mAddress=131,mAttributes=3,mMaxPacketSize=16,mInterval=8]]
    UsbInterface[mId=3,mAlternateSetting=0,mName=Serial port bulk data,mClass=10,mSubclass=0,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=130,mAttributes=2,mMaxPacketSize=512,mInterval=0]
[... 57 lines cut ...]
2022-01-17 16:54:53.386 650-737/? E/ExtCamDev@3.4: trimSupportedFormats: input format list is empty!
2022-01-17 16:54:53.387 650-737/? I/CamPvdr@2.4-external: ExtCam: adding /dev/video5 to External Camera HAL!
2022-01-17 16:54:53.388 1296-1465/? D/UsbDeviceDescriptor:   1 configs
2022-01-17 16:54:53.392 1296-1465/? D/UsbHostManager: Added device UsbDevice[mName=/dev/bus/usb/001/004,mVendorId=2507,mProductId=16391,mClass=239,mSubclass=2,mProtocol=1,mManufacturerName=FLIR,mProductName=Boson,mVersion=1.00,mSerialNumber=58283,mConfigurations=[
    UsbConfiguration[mId=1,mName=null,mAttributes=192,mMaxPower=250,mInterfaces=[
    UsbInterface[mId=0,mAlternateSetting=0,mName=FLIR Camera,mClass=14,mSubclass=1,mProtocol=0,mEndpoints=[]
    UsbInterface[mId=1,mAlternateSetting=0,mName=Video In (disabled),mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=129,mAttributes=2,mMaxPacketSize=512,mInterval=0]]
    UsbInterface[mId=2,mAlternateSetting=0,mName=Serial port interface,mClass=2,mSubclass=2,mProtocol=1,mEndpoints=[
    UsbEndpoint[mAddress=131,mAttributes=3,mMaxPacketSize=16,mInterval=8]]
    UsbInterface[mId=3,mAlternateSetting=0,mName=Serial port bulk data,mClass=10,mSubclass=0,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=130,mAttributes=2,mMaxPacketSize=512,mInterval=0]
    UsbEndpoint[mAddress=1,mAttributes=2,mMaxPacketSize=512,mInterval=0]]]]
2022-01-17 16:54:53.394 972-1252/? I/CameraProviderManager: Enumerating new camera device: device@3.4/external//dev/video5
2022-01-17 16:54:53.394 650-737/? E/ExtCamDev@3.4: getFrameRateList: failed to get supported frame rates for format:YU12 w 640 h 512
2022-01-17 16:54:53.395 650-737/? E/ExtCamDev@3.4: trimSupportedFormats: input format list is empty!
2022-01-17 16:54:53.397 972-1252/? I/CameraService: onDeviceStatusChanged: Status changed for cameraId=/dev/video5, newStatus=1
2022-01-17 16:54:53.397 972-1252/? I/CameraService: onDeviceStatusChanged: Unknown camera ID /dev/video5, a new camera is added
2022-01-17 16:54:53.397 972-1252/? W/CameraFlashlight: hasFlashUnitLocked: camera /dev/video5 not present when findFlashUnits() was called
2022-01-17 16:54:53.397 1296-1319/? D/UsbDeviceManager: Clear notification
2022-01-17 16:54:53.402 1296-1313/? D/CompatibilityInfo: mCompatibilityFlags - 4
2022-01-17 16:54:53.402 1296-1313/? D/CompatibilityInfo: applicationDensity - 240
2022-01-17 16:54:53.402 1296-1313/? D/CompatibilityInfo: applicationScale - 1.0
```
Could anyone offer a tip on how to look into this to find a solution?

lieberk2-hd, 2023-08-23:
Please let me know if you ever got this figured out.  I am struggling with FLIR cameras right now.

Clark-caipeiyuan, 2023-08-23:
收到了，谢谢！

zipswich, 2023-08-23:
@lieberk2-hd No, unfortunately. Hope you will be able to find a solution and share it here.