saki4510t/UVCCamera #643: A/libc: Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x0 in tid 19263 (CameraThread), pid 18571 (example.mundial)

TetrixGauss, 2022-02-15:
Hello! In Samsung galaxy Tab A7 tablet i get A/libc: Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x0 in tid 19263 (CameraThread), pid 18571 (example.mundial) that error and no other logs and the app crashes when i disconnect the usb from the tablet! 

Can someone please help me either with the error if you know what is or the way to get more logs out of the library for the error! 

I know that is a null pointer dereference but i dont know where!

I have tryied blindly to check for NULLs inside the code but of'course with no effect!
Please keep in mind that in other tablets the above scenario plays well but on the Samsung Galaxy Tab A7 crashes!

Please any help is appreciated!

ExcailburWei, 2022-02-16:
[https://www.jianshu.com/p/972e05fa76a3](url)

ExcailburWei, 2022-02-16:
https://www.jianshu.com/p/972e05fa76a3

TetrixGauss, 2022-02-16:
@ExcailburWei Thank you so much! Now the app doesn't crash but i cannot do anything and it tells me "Stop app or Wait".
I have no errors in the console. Any suggestions?

Thank you so much in advance!