saki4510t/UVCCamera #657: is it possible to push two streams(video) simultainously with different resolution(640x480 and 1920x1080)?

1goresha, 2022-04-10:
i want to push simultainously one stream for udp transmitting and another for recording, and it would be great if it will be without preview.

1goresha, 2022-04-10:
i want to push simultainously one stream for udp transmitting and another for recording, and it would be great if it will be without preview.
anybody has ideas how to do it, please?

9ghtX, 2022-04-12:
I've done this with two cameras. It's possible, just create a new instance of camera and view an image on different ImageView in layout

9ghtX, 2022-04-13:
> > I've done this with two cameras. It's possible, just create a new instance of camera and view an image on different ImageView in layout
> 
> thank you for reply, but i need to get bytes of encoded data(H264) from there, how can i do this?

Unfortunately, I don't know. I opened my own Issue post about this(https://github.com/saki4510t/UVCCamera/issues/658).