saki4510t/UVCCamera #660: Where does the preview delay mainly come from?

zipswich, 2022-04-27:
The delay for resolution 4656x3496 is about 1000ms.
I took a look at do_preview void UVCPreview::do_preview(uvc_stream_ctrl_t *ctrl) in UVCPreview.cpp.

		for ( ; LIKELY(isRunning()) ; ) {
				frame_mjpeg = waitPreviewFrame();
				if (LIKELY(frame_mjpeg)) {
					//The following takes no time
					frame = get_frame(frame_mjpeg->width * frame_mjpeg->height * 2);
        			       //The following takes 150ms
        			       result = uvc_mjpeg2yuyv(frame_mjpeg, frame);   // MJPEG => yuyv
					//The following takes no time
					recycle_frame(frame_mjpeg);
					if (LIKELY(!result)) {
			                       //The following takes 200 ms for uvc_mjpeg2yuyv(frame_mjpeg, frame)
						frame = draw_preview_one(frame, &mPreviewWindow, uvc_any2rgbx, 4);
			                       //The entire process takes 350ms for uvc_mjpeg2yuyv(frame_mjpeg, frame)
						addCaptureFrame(frame);
					} else {
						recycle_frame(frame);
					}
				}
			}

Each iteration in the loop takes about 350ms, so the major part of the 1000ms delay is from somewhere else. Could anyone offer a tip on this?

liu408309046, 2022-04-28:
Is there any solution? I have encountered a similar problem. When the resolution is set to a large one, the preview delay is very high

shengjunhu, 2022-06-09:
> Is there any solution? I have encountered a similar problem. When the resolution is set to a large one, the preview delay is very high

What is the maximum resolution you use ? Decoding is time-consuming during rendering.

zhouqunhuang, 2024-05-11:
如何修改可以降低延迟。这延迟真的很头疼！！！

zhouqunhuang, 2024-05-11:
How to modify to reduce latency. This delay is really a headache!!

atulsingh1204, 2025-08-26:
You can use tjDecompress2()

xiao923700243, 2025-10-31:
You should retrieve the RAW data in the callback and then call Android's decoding module to decode and display it.

Clark-caipeiyuan, 2025-10-31:
收到了，谢谢！