saki4510t/UVCCamera #669: How to captureStill image with camera.captureStillImage?

Christina-ko, 2022-08-17:
Hi, 

Currently I am working to capture stillCapture in order to push to AI machine to read OCR. 
In order to get live preview while getting the trigger button signal, I can only use the uvcCamera library. When I implement captureStill function from AbstractUVCCameraHandler, my apps will crash and stop. Anyone face this issue before? Can give me some guide? Really appreciate on your help. 

  @Override
        public void onConnectDev ( final UsbDevice device, final USBMonitor.UsbControlBlock ctrlBlock, final boolean isConnected ) {
            if (!isConnected) {
                showShortMsg ( "fail to connect,please check resolution params" );
                isPreview = false;
            } else {
                isPreview = true;
                showShortMsg ( "Camera is connected" );
                final UVCCamera camera = new UVCCamera ( );
                camera.open ( ctrlBlock );
                camera.setPreviewSize ( 1280, 720, FRAME_FORMAT_MJPEG );
                camera.setAutoWhiteBlance ( false );
                camera.updateCameraParams ( );
                final SurfaceTexture st = mUVCCameraView.getSurfaceTexture ( );
                if (st != null) {
                    mPreviewSurface = new Surface ( st );
                    camera.setPreviewDisplay ( mPreviewSurface );
                    camera.startPreview ( );

                }

                    queueEvent( new Runnable ( ) {
                    @Override
                    public void run ( ) {
                     camera.setButtonCallback ( new IButtonCallback ( ) {
                            @Override
                            public void onButton ( final int button, final int state ) {

                                runOnUiThread ( new Runnable ( ) {
                                    @Override
                                    public void run ( ) {
                                        final Toast toast = Toast.makeText (USBCameraActivity.this, "onButton(button=" + button + "; " +
                                                "state=" + state + ")", Toast.LENGTH_SHORT );
                                        toast.show ( );
                                        if (button ==1 && state ==1){
                                            toast.show ( );
                                            camera.setFrameCallback ( mIFrameCallback, UVCCamera.PIXEL_FORMAT_RGB565 );

                                        }

                                    }
                                } );
                            }
                        } );

                    }

                } );

            }
        }

 private final IFrameCallback mIFrameCallback = new IFrameCallback ( ) {
        @Override
        public void onFrame ( ByteBuffer frame ) {
            synchronized (mSync){
                if (tempBitmap == null){
                    tempBitmap = Bitmap.createBitmap ( 1280, 720, Bitmap.Config.RGB_565 );
                }
                frame.clear ();
                tempBitmap.copyPixelsFromBuffer ( frame );
            }
            mResultView.post ( mUpdateTask );
        }
    };
    private Runnable mUpdateTask = new Runnable ( ) {
        @Override
        public void run ( ) {
            synchronized (mSync){
                mResultView.setImageBitmap ( tempBitmap );
            }
        }
    };

Keerthana-N, 2023-03-06:
Hi, did you get any solution?

Christina-ko, 2023-03-06:
Yes. I found another uvc library. https://github.com/shiyinghan/UVCAndroid

Keerthana-N, 2023-03-06:
Were you able to get just still image without previewing? Actually I am trying to just capture still image without streaming & previewing. Can you please suggest any suitable library if you have tried.