saki4510t/UVCCamera #671: Latency in Image preview

asadasivam, 2022-09-08:
Does latency in Image preview could be reduced by showing Bitmap directly in Preview Area, instead of saving it as compressed image (.PNG) ?

asadasivam, 2022-09-08:
And, why compressed for saving bitmap as .PNG file

asadasivam, 2022-09-08:
Pls need a quick support on this issue.

shengjunhu, 2022-09-09:
> Does latency when showing the .png file to the image preview can be reduced by showing Bitmap directly?

Image decoding and image format conversion are time-consuming, and the larger the image size, the more time-consuming it is

asadasivam, 2022-09-09:
Sry my question does n't asked in a correct way. 

Use case : _Button Press -> Capture current frame and display in Image preview area implementing using 480P camera_

Code flow: 
1. button press -> calling captureStill method in _AbstractUVCCameraHandler class_
2. In handleCaptureStill method-> Capturing current frame as a bitmap and save it in a compressed file(.PNG). 
3. Using setImageBitmap(_ImageView.class_) and bringToFront(_View.class_) methods, we direct the compressed file into Image view

My question is can't we directly show the _bitmap in Image preview area_ instead of 
encoding into a file and then show the file in Image preview area?

shengjunhu, 2022-09-14:
> Sry my question does n't asked in a correct way.
>
> Use case : _Button Press -> Capture current frame and display in Image preview area implementing using 480P camera_
>
> Code flow:
>
> 1. button press -> [...]

It is not recommended to use this method. It will be inefficient, and the CPU and memory will shake severely, which will also cause the preview to get stuck

asadasivam, 2022-09-14:
> It is not recommended to use this method. It will be inefficient, and the CPU and memory will shake severely, which will also cause the preview to get stuck

- From the IFrameCallback.java, we understood another possible. Does its a recommended method?
"If you want to get same result as on screen, please try to
	 * consider to get images via texture(SurfaceTexture) and read pixel buffer from it using OpenGL|ES2/3
	 * instead of using IFrameCallback(this way is much efficient in most case than using IFrameCallback)"

1729051657, 2024-01-19:
> > It is not recommended to use this method. It will be inefficient, and the CPU and memory will shake severely, which will also cause the preview to get stuck不建议使用此方法。效率会很低，CPU和内存会剧烈抖动，也会导致预览卡住
> 
> * From the IFrameCallback.java, we understood another possible. Does its a recommended method?从 IFrameCallback.java 中，我们理解了另一种可能性。这是推荐的方法吗？
>   "If you want to get same result as on screen, please try to
>   “如果您想获得与屏幕上相同的结果，请尝试
>   * consider to get images via texture(SurfaceTexture) and read pixel buffer from it using OpenGL|ES2/3考虑通过 texture（SurfaceTexture） 获取图像并使用 OpenGL|ES2/3 [美]
>   * instead of using IFrameCallback(this way is much efficient in most case than using IFrameCallback)"而不是使用 IFrameCallback（这种方式在大多数情况下比使用 IFrameCallback 要有效得多）”

如何实现呢？