saki4510t/UVCCamera #672: Preview Delay

yangdefu168, 2022-09-16:
2592X1944
void UVCPreview::addPreviewFrame(uvc_frame_t *frame) {

    pthread_mutex_lock(&preview_mutex);
    if (isRunning() && (previewFrames.size() < MAX_FRAME)) {
        previewFrames.clear();//clear previewFrames
        previewFrames.put(frame);
        frame = NULL;
        pthread_cond_signal(&preview_sync);
    }
    pthread_mutex_unlock(&preview_mutex);
    if (frame) {
        recycle_frame(frame);
    }
}

add previewFrames.clear(); Preview Smooth。。
There are still 200 delays

How to preview normally

yangdefu168, 2022-09-16:
Please take a look. Thank you very much

zhouqunhuang, 2024-05-11:
Hello! Have you solved it? If you have, could you please tell me how you fixed it?