saki4510t/UVCCamera #674: USBMonitor的问题还是抽时间改下吧

annaijinwind, 2022-09-28:
`mPermissionIntent = PendingIntent.getBroadcast(context, 0, new Intent(ACTION_USB_PERMISSION), PendingIntent.FLAG_IMMUTABLE);`
就这行，之前有老哥反应把PendingIntent.FLAG_MUTABLE改为PendingIntent.FLAG_MUTABLE，我试了下确实可以，在ap31以上的机器，如果用原来的写法，会出现第一次启动获取权限失败并且需要反复拔插重启后才能使用的问题，你可以改下试试
[https://github.com/jiangdongguo/AndroidUSBCamera/issues/434#issuecomment-1250231199](url)

l376571926, 2023-04-14:
> `mPermissionIntent = PendingIntent.getBroadcast(context, 0, new Intent(ACTION_USB_PERMISSION), PendingIntent.FLAG_IMMUTABLE);` [...]

这人是日本人，你说中文他怕是看不懂

jooowa, 2023-04-14:
ap31以上的同意後沒有直接顯示畫面 這個有解法了嗎