saki4510t/UVCCamera #687: pthread_mutex_lock called on a destroyed mutex

NanayaHaruki, 2023-06-29:
stream.c
_uvc_delete_transfer
pthread_mutex_lock(&strmh->cb_mutex);	// XXX crash while calling uvc_stop_streaming

I found that you moved the free transfer buffer from `uvc_stream_stop` to `_uvc_stream_callback` to avoid the crash. However, when the `_uvc_delete_transfer` function is invoked, the `cb_mutex` has already been destroyed. 
Therefore, the line `pthread_mutex_lock(&strmh->cb_mutex);` will cause a crash.
How to solve this problem? 

```
A  signal 6 (SIGABRT), code -1 (SI_QUEUE), fault addr --------
A  Abort message: 'FORTIFY: pthread_mutex_lock called on a destroyed mutex (0xb400006e8a9c1920)'
A      x0  0000000000000000  x1  00000000000030b1  x2  0000000000000006  x3  0000006d1e747710
A      x4  0000000000000000  x5  0000000000000000  x6  0000000000000000  x7  0000000000000028
A      x8  00000000000000f0  x9  000000703d77a7c0  x10 ffffff80fffffbdf  x11 0000000000000001
A      x12 0000000000000020  x13 0000005e6b2d7ec7  x14 000a78e2fa845ed6  x15 0000000029aaaaf1
A      x16 000000703d812c80  x17 000000703d7f4320  x18 0000006d16500000  x19 0000000000003015
A      x20 00000000000030b1  x21 00000000ffffffff  x22 b400006e8a9c1870  x23 b400006faab7a200
A      x24 b400006e0a98a110  x25 b400006e8a9b6d50  x26 b400006e4a9b8d70  x27 0000000000000002
A      x28 b400006e0a98a110  x29 0000006d1e747790
A      lr  000000703d7a8148  sp  0000006d1e7476f0  pc  000000703d7a8178  pst 0000000000001000
A  backtrace:
A        #00 pc 000000000004e178  /apex/com.android.runtime/lib64/bionic/libc.so (abort+168) (BuildId: 
A        #01 pc 00000000000b0e74  /apex/com.android.runtime/lib64/bionic/libc.so (__fortify_fatal(char const*, ...)
A        #02 pc 00000000000b0504  /apex/com.android.runtime/lib64/bionic/libc.so (HandleUsingDestroyedMutex
A        #03 pc 00000000000b039c  /apex/com.android.runtime/lib64/bionic/libc.so (pthread_mutex_lock+192) (BuildId: 
A        #04 pc 00000000000156f4  /data/app/~~QFV7DXoKVdaeJVFSiKlcmA==/com.optimams.31f851734c2b87b2fdbfdc6a39378593)
A        #05 pc 0000000000014d34  /data/app/~~QFV7DXoKVdaeJVFSiKlcmA==/com.optimams.31f851734c2b87b2fdbfdc6a39378593)
A        #06 pc 00000000000105a8  /data/app/~~QFV7DXoKVdaeJVFSiKlcmA==/com.optimams.ansfer_completion+328) (BuildId: 3ff7f2c27cfec84f0f301cf626ffe08df8be9b1a)
A        #07 pc 000000000001063c  /data/app/~~QFV7DXoKVdaeJVFSiKlcmA==/com.optimams.ansfer_cancellation+32) (BuildId: 3ff7f2c27cfec84f0f301cf626ffe08df8be9b1a)
A        #08 pc 0000000000015964  /data/app/~~QFV7DXoKVdaeJVFSiKlcmA==/com.optimams.2c27cfec84f0f301cf626ffe08df8be9b1a)
A        #09 pc 00000000000110e8  /data/app/~~QFV7DXoKVdaeJVFSiKlcmA==/com.optimams.2c27cfec84f0f301cf626ffe08df8be9b1a)
A        #10 pc 0000000000010bd0  /data/app/~~QFV7DXoKVdaeJVFSiKlcmA==/com.optimams.events_timeout_completed+432) (BuildId: 3ff7f2c27cfec84f0f301cf626ffe08df8be9b1a)
A        #11 pc 00000000000112a4  /data/app/~~QFV7DXoKVdaeJVFSiKlcmA==/com.optimams.events+48) (BuildId: 3ff7f2c27cfec84f0f301cf626ffe08df8be9b1a)
A        #12 pc 0000000000013734  /data/app/~~QFV7DXoKVdaeJVFSiKlcmA==/com.optimams.s+132) (BuildId: 3efaa13c31f851734c2b87b2fdbfdc6a39378593)
A        #13 pc 00000000000af880  /apex/com.android.runtime/lib64/bionic/libc.so (__pthread_start(void*)+64) 
A        #14 pc 00000000000500d0  /apex/com.android.runtime/lib64/bionic/libc.so (__start_thread+64) (BuildId: 
```

aaron23v, 2025-03-25:
Did you find a solution for this?

NanayaHaruki, 2025-03-26:
> [@NanayaHaruki](https://github.com/NanayaHaruki) Did you find a solution for this?

Nope, I switched to using V4L2 to control my camera based on this [https://github.com/shengjunhu/android_camera_v4l2].

biaji, 2025-04-28:
看看  https://github.com/saki4510t/UVCCamera/pull/591