saki4510t/UVCCamera #700: Android11, RK3568, null pointer dereference

longxinpharma, 2024-03-30:
*** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
Build fingerprint: 'rockchip/rk3568_r/rk3568_r:11/RQ3A.210705.001/eng.topeet.20230922.103357:userdebug/release-keys'
Revision: '0'
ABI: 'arm64'
Timestamp: 2024-03-30 08:49:14+0800
pid: 8112, tid: 8580, name: Thread-107  >>> com.besberry.fraction <<<
uid: 1000
signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x7f8
Cause: null pointer dereference
    x0  0000000000000000  x1  0000000000000004  x2  0000006f47d98d8f  x3  0000006f47d98dbd
    x4  0000000000000001  x5  0000000000000000  x6  0000000000000ff0  x7  7ca47ea37ca880ae
    x8  00000000000007f8  x9  0000000000000001  x10 0000000000000000  x11 0000006f5c99ba60
    x12 ffffff80ffffffe0  x13 0000006f47d9c6c8  x14 0018538c58b7c6d8  x15 0000000029aaaaf1
    x16 0000006f47d9cea0  x17 000000725fd0d2dc  x18 0000006f34d78000  x19 0000000000000002
    x20 b4000070ad614744  x21 000000000000000c  x22 0000000000000bf4  x23 0000000000000000
    x24 b4000070ad614930  x25 b40000702d5a4290  x26 b400006fed611c48  x27 b40000701d5a64f0
    x28 0000000000000bf4  x29 0000006f5c99bb20
    lr  0000006f47d9329c  sp  0000006f5c99ba60  pc  0000006f47d932b0  pst 0000000080001000

backtrace:
      #00 pc 00000000000102b0  /data/app/~~H7eI_zkNqTJfWRAbk5nXXA==/com.besberry.fraction-ioChKHiNxkZgcv9FD6qAWg==/base.apk!libusb100.so (offset 0x147de000) (BuildId: 05c0d08a1cf7f7fc833ce4a853ebd4160c2f0a77)
      #01 pc 000000000000c014  /data/app/~~H7eI_zkNqTJfWRAbk5nXXA==/com.besberry.fraction-ioChKHiNxkZgcv9FD6qAWg==/base.apk!libusb100.so (offset 0x147de000) (BuildId: 05c0d08a1cf7f7fc833ce4a853ebd4160c2f0a77)
      #02 pc 000000000000ba5c  /data/app/~~H7eI_zkNqTJfWRAbk5nXXA==/com.besberry.fraction-ioChKHiNxkZgcv9FD6qAWg==/base.apk!libusb100.so (offset 0x147de000) (libusb_handle_events_timeout_completed+600) (BuildId: 05c0d08a1cf7f7fc833ce4a853ebd4160c2f0a77)
      #03 pc 000000000000c174  /data/app/~~H7eI_zkNqTJfWRAbk5nXXA==/com.besberry.fraction-ioChKHiNxkZgcv9FD6qAWg==/base.apk!libusb100.so (offset 0x147de000) (libusb_handle_events+32) (BuildId: 05c0d08a1cf7f7fc833ce4a853ebd4160c2f0a77)
      #04 pc 0000000000010cd8  /data/app/~~H7eI_zkNqTJfWRAbk5nXXA==/com.besberry.fraction-ioChKHiNxkZgcv9FD6qAWg==/base.apk!libuvc.so (offset 0x264b000) (_uvc_handle_events+132) (BuildId: 9c3a9fab87f4bd98ce4335bab9b57b984b16f59c)
      #05 pc 00000000000af880  /apex/com.android.runtime/lib64/bionic/libc.so (__pthread_start(void*)+64) (BuildId: 23b47e338fca2b85af1ec50fd590d7c7)
      #06 pc 00000000000500d0  /apex/com.android.runtime/lib64/bionic/libc.so (__start_thread+64) (BuildId: 23b47e338fca2b85af1ec50fd590d7c7)

Clark-caipeiyuan, 2024-03-30:
收到了，谢谢！

thunder3568, 2024-04-24:
I have a same issue on [rk3568 android11]