saki4510t/UVCCamera #706: Execution failed for task ':libuvc:ndkBuild'.

MrAdi149, 2024-06-25:
i tried your Android Usb Camera project but i got erron on ndkBuild, please help me to solve the issue

my local.properties file look like this:

sdk.dir=C\:\\Users\\EXT.USER\\AppData\\Local\\Android\\Sdk
#ndk.dir=C\:\\Users\\EXT.USER\\AppData\\Local\\Android\\Sdk\\ndk\\27.0.11902837

my build.gradle(:libuvc) look like this:

apply plugin: 'com.android.library'

import org.apache.tools.ant.taskdefs.condition.Os

android {
	namespace = "com.jiangdg.uvccamera"
	buildToolsVersion = "34.0.0"
	compileSdk = 35

	compileOptions {
		sourceCompatibility = JavaVersion.VERSION_1_8
		targetCompatibility = JavaVersion.VERSION_1_8
	}

	defaultConfig {
		minSdk = 24
		targetSdk = 34
	}

	lintOptions {
		checkReleaseBuilds false
		abortOnError false
		disable 'MissingTranslation'
	}

	ndkVersion '27.0.11902837'

	buildTypes {
		release {
			minifyEnabled false
			proguardFiles getDefaultProguardFile('proguard-android.txt'), 'proguard-rules.txt'
		}
	}
}

tasks.withType(JavaCompile) {
	compileTask -> compileTask.dependsOn ndkBuild
}

String getNdkBuildPath() {
	Properties properties = new Properties()
	properties.load(project.rootProject.file('local.properties').newDataInputStream())
	def ndkBuildingDir = properties.getProperty("ndk.dir")
	def ndkBuildPath = ndkBuildingDir
	if (Os.isFamily(Os.FAMILY_WINDOWS)) {
		ndkBuildPath = ndkBuildingDir + '/ndk-build.cmd'
	} else {
		ndkBuildPath = ndkBuildingDir + '/ndk-build'
	}
	return ndkBuildPath
}

task ndkBuild(type: Exec, description: 'Compile JNI source via NDK') {
	println('executing ndkBuild')
	def ndkBuildPath = getNdkBuildPath();
	commandLine ndkBuildPath, '-j8', '-C', file('src/main').absolutePath
}

task ndkClean(type: Exec, description: 'clean JNI libraries') {
	println('executing ndkBuild clean')
	def ndkBuildPath = getNdkBuildPath();
	commandLine ndkBuildPath, 'clean', '-C', file('src/main').absolutePath
}

clean.dependsOn 'ndkClean'

dependencies {
	implementation fileTree(dir: new File(buildDir, 'libs'), include: '*.jar')
	implementation "androidx.appcompat:appcompat:1.7.0"
	implementation 'com.elvishew:xlog:1.11.0'
//	implementation project(path: ':libuvccommon')
}

now i am getting error like this:

Execution failed for task ':libuvc:ndkBuild'.
> A problem occurred starting process 'command 'null/ndk-build.cmd''

* Try:
> Run with --scan to get full insights.
> Get more help at https://help.gradle.org.

* Exception is:
org.gradle.api.tasks.TaskExecutionException: Execution failed for task ':libuvc:ndkBuild'.
at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter.lambda$executeIfValid$1(ExecuteActionsTaskExecuter.java:130)
at org.gradle.internal.Try$Failure.ifSuccessfulOrElse(Try.java:293)
at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter.executeIfValid(ExecuteActionsTaskExecuter.java:128)
at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter.execute(ExecuteActionsTaskExecuter.java:116)
at org.gradle.api.internal.tasks.execution.FinalizePropertiesTaskExecuter.execute(FinalizePropertiesTaskExecuter.java:46)
at org.gradle.api.internal.tasks.execution.ResolveTaskExecutionModeExecuter.execute(ResolveTaskExecutionModeExecuter.java:51)
at org.gradle.api.internal.tasks.execution.SkipTaskWithNoActionsExecuter.execute(SkipTaskWithNoActionsExecuter.java:57)
at org.gradle.api.internal.tasks.execution.SkipOnlyIfTaskExecuter.execute(SkipOnlyIfTaskExecuter.java:74)
at org.gradle.api.internal.tasks.execution.CatchExceptionTaskExecuter.execute(CatchExceptionTaskExecuter.java:36)
at org.gradle.api.internal.tasks.execution.EventFiringTaskExecuter$1.executeTask(EventFiringTaskExecuter.java:77)
at org.gradle.api.internal.tasks.execution.EventFiringTaskExecuter$1.call(EventFiringTaskExecuter.java:55)
at org.gradle.api.internal.tasks.execution.EventFiringTaskExecuter$1.call(EventFiringTaskExecuter.java:52)
at org.gradle.internal.operations.DefaultBuildOperationRunner$CallableBuildOperationWorker.execute(DefaultBuildOperationRunner.java:209)
at org.gradle.internal.operations.DefaultBuildOperationRunner$CallableBuildOperationWorker.execute(DefaultBuildOperationRunner.java:204)
at org.gradle.internal.operations.DefaultBuildOperationRunner$2.execute(DefaultBuildOperationRunner.java:66)
at org.gradle.internal.operations.DefaultBuildOperationRunner$2.execute(DefaultBuildOperationRunner.java:59)
at org.gradle.internal.operations.DefaultBuildOperationRunner.execute(DefaultBuildOperationRunner.java:166)
at org.gradle.internal.operations.DefaultBuildOperationRunner.execute(DefaultBuildOperationRunner.java:59)
at org.gradle.internal.operations.DefaultBuildOperationRunner.call(DefaultBuildOperationRunner.java:53)
at org.gradle.api.internal.tasks.execution.EventFiringTaskExecuter.execute(EventFiringTaskExecuter.java:52)
at org.gradle.execution.plan.LocalTaskNodeExecutor.execute(LocalTaskNodeExecutor.java:42)
at org.gradle.execution.taskgraph.DefaultTaskExecutionGraph$InvokeNodeExecutorsAction.execute(DefaultTaskExecutionGraph.java:331)
at org.gradle.execution.taskgraph.DefaultTaskExecutionGraph$InvokeNodeExecutorsAction.execute(DefaultTaskExecutionGraph.java:318)
at org.gradle.execution.taskgraph.DefaultTaskExecutionGraph$BuildOperationAwareExecutionAction.lambda$execute$0(DefaultTaskExecutionGraph.java:314)
at org.gradle.internal.operations.CurrentBuildOperationRef.with(CurrentBuildOperationRef.java:85)
at org.gradle.execution.taskgraph.DefaultTaskExecutionGraph$BuildOperationAwareExecutionAction.execute(DefaultTaskExecutionGraph.java:314)
at org.gradle.execution.taskgraph.DefaultTaskExecutionGraph$BuildOperationAwareExecutionAction.execute(DefaultTaskExecutionGraph.java:303)
at org.gradle.execution.plan.DefaultPlanExecutor$ExecutorWorker.execute(DefaultPlanExecutor.java:459)
at org.gradle.execution.plan.DefaultPlanExecutor$ExecutorWorker.run(DefaultPlanExecutor.java:376)
at org.gradle.internal.concurrent.ExecutorPolicy$CatchAndRecordFailures.onExecute(ExecutorPolicy.java:64)
at org.gradle.internal.concurrent.AbstractManagedExecutor$1.run(AbstractManagedExecutor.java:48)
Caused by: org.gradle.process.internal.ExecException: A problem occurred starting process 'command 'null/ndk-build.cmd''
at org.gradle.process.internal.DefaultExecHandle.execExceptionFor(DefaultExecHandle.java:241)
at org.gradle.process.internal.DefaultExecHandle.setEndStateInfo(DefaultExecHandle.java:218)
at org.gradle.process.internal.DefaultExecHandle.failed(DefaultExecHandle.java:396)
at org.gradle.process.internal.ExecHandleRunner.lambda$run$3(ExecHandleRunner.java:102)
at org.gradle.internal.operations.CurrentBuildOperationRef.with(CurrentBuildOperationRef.java:85)
at org.gradle.process.internal.ExecHandleRunner.run(ExecHandleRunner.java:101)
at org.gradle.internal.concurrent.ExecutorPolicy$CatchAndRecordFailures.onExecute(ExecutorPolicy.java:64)
at org.gradle.internal.concurrent.AbstractManagedExecutor$1.run(AbstractManagedExecutor.java:48)
Caused by: net.rubygrapefruit.platform.NativeException: Could not start 'null/ndk-build.cmd'

Clark-caipeiyuan, 2024-06-25:
收到了，谢谢！

MrAdi149, 2024-06-25:
请帮我解决这个问题，这真的很重要