saki4510t/UVCCamera #707: Preview last frame when external camera is removed.

aaron23v, 2024-07-30:
Just wanted to know, how can we implement this?

When the external camera is detached / removed, i want to preview the last frame and when it is attached back, it should start working as normal. Need some guidance

Clark-caipeiyuan, 2024-07-30:
收到了，谢谢！

fbzhong, 2024-07-30:
I would suggest you to use ImageReader & ImageWriter as a proxy between UVCCamera and Surface.

Set ImageReader as UVCCamera set preview display. And then enqueue to ImageWriter in onImageAvailable listener.

aaron23v, 2025-01-16:
> I would suggest you to use ImageReader & ImageWriter as a proxy between UVCCamera and Surface.
> 
> Set ImageReader as UVCCamera set preview display. And then enqueue to ImageWriter in onImageAvailable listener. Then you could always keep the last Image (into Bitmap)

Thanks for this advice. Will check it out

Clark-caipeiyuan, 2025-01-16:
收到了，谢谢！

AaronDsilva97, 2025-01-20:
Hi, could you please clarify in which file or folder I should add the ImageReader and image queuing logic in the UVCCamera project?