saki4510t/UVCCamera #713: 3.2.10版本 call stopPreview會crash

jenny850113, 2024-10-28:
使用3.2.10進行測試,發現call了stoppreview後會出現這個exception
pid: 15503, tid: 19741, name: camera_manager  >>>
#01 pc 0000000000018b38  /data/app/~~naPIxY-21gO5SIdGIKrLxg==/com.benq.qshare.client-Ktu34aU_hUaQwDqyn28UPw==/base.apk!libUVCCamera.so (copyToSurface(uvc_frame*, ANativeWindow**)+236) (BuildId: 5d40a904943d54cb55ccde8773b95cefe8cd148f)
pc 00000000000185d4  /data/app/~~naPIxY-21gO5SIdGIKrLxg==/com.benq.qshare.client-Ktu34aU_hUaQwDqyn28UPw==/base.apk!libUVCCamera.so (UVCPreview::do_preview(uvc_stream_ctrl*)+488) (BuildId: 5d40a904943d54cb55ccde8773b95cefe8cd148f)
#03 pc 0000000000017d24  /data/app/~~naPIxY-21gO5SIdGIKrLxg==/com.benq.qshare.client-Ktu34aU_hUaQwDqyn28UPw==/base.apk!libUVCCamera.so (UVCPreview::preview_thread_func(void*)+44) (BuildId: 5d40a904943d54cb55ccde8773b95cefe8cd148f)

biaji, 2025-04-28:
是不是这个：https://github.com/saki4510t/UVCCamera/pull/520

Clark-caipeiyuan, 2025-04-28:
收到了，谢谢！

jenny850113, 2025-04-28:
it will still crash in 3.3.3

jenny850113, 2025-04-28:
It will still crash in v3.3.3