saki4510t/UVCCamera #718: How to Change the Background to Black When the USB Camera is Disconnected?

stopkite, 2025-02-05:
When the USB camera is disconnected, the screen appears **white** due to the absence of a video feed. However, setting `backgroundColor` to **black** does not seem to work. Is there a way to change the background to black in this situation?

shengjunhu, 2025-02-05:
> When the USB camera is disconnected, the screen appears **white** due to the absence of a video feed. However, setting `backgroundColor` to **black** does not seem to work. Is there a way to change the background to black in this situation?
> 
> Do I need to modify the C code inside the library to fix this?

When the USB camera is disconnected, clear the surface content：memset(buffer.bits)
[UVCPreview.cpp::copyToSurface](https://github.com/saki4510t/UVCCamera/blob/master/libuvccamera/src/main/jni/UVCCamera/UVCPreview.cpp#L596)

stopkite, 2025-02-05:
@shengjunhu  
First of all, thank you for your response!
I tried changing the value in memset (0, 100, 255, 16, etc.), but none of them worked as expected.

**ex)** `memset(dest, 100, buffer.height * dest_step);`

As a temporary solution, I wrapped the **SurfaceView** inside a **FrameLayout** and placed a black overlay on top of it. However, since this approach adds an extra view, I believe it consumes slightly more resources compared to modifying the C++ side to render a black screen directly.

If anyone manages to solve this issue, I would greatly appreciate it if you could share the solution. Thank you!" 🚀

stopkite, 2025-02-05:
## ✅ Solution
When using `memset()` to set RGBA values, an issue was encountered where the **A (Alpha) value was not applied correctly.**
This is suspected to be because `memset()` operates **on a byte-by-byte basis,** making it unable to properly set the RGBA (4-byte) structure.

To resolve this, the **Alpha value was explicitly set to 255 on a per-pixel basis** to ensure an opaque background.
Additionally, modifications were made to the `clearDisplay()` function, where `memset()` is actually used, to correctly apply the intended values.

### 💻 Updated Code
```cpp
void UVCPreview::clearDisplay() {
    ENTER();

    ANativeWindow_Buffer buffer;
    pthread_mutex_lock(&capture_mutex);
    {
        if (LIKELY(mCaptureWindow)) {
                      // code omitted
        }
    }
    pthread_mutex_unlock(&capture_mutex);

    pthread_mutex_lock(&preview_mutex);
    {
        if (LIKELY(mPreviewWindow)) {
            if (LIKELY(ANativeWindow_lock(mPreviewWindow, &buffer, NULL) == 0)) {
                uint8_t *dest = (uint8_t *)buffer.bits;
                const int stride = buffer.stride * PREVIEW_PIXEL_BYTES;
           
                for (int i = 0; i < buffer.height; i++) {
                    uint8_t *row = dest;
                    for (int j = 0; j < buffer.width; j++) {
                        row[0] = 0;    // R (Red)
                        row[1] = 0;    // G (Green)
                        row[2] = 0;    // B (Blue)
                        row[3] = 255;  // A (Alpha, fully opaque)
                        row += 4; // Move to the next pixel
                    }
                    dest += stride;
                }

                ANativeWindow_unlockAndPost(mPreviewWindow);
            }
        }
    }
    pthread_mutex_unlock(&preview_mutex);

    EXIT();
}
```

closed 2025-02-05 by stopkite

shengjunhu, 2025-02-06:
> ## ✅ Solution
> 用于设置 RGBA 值时，遇到 **A （Alpha） 值未正确应用的问题。**这可能是因为**逐字节运行，**使其无法正确设置 RGBA（4 字节）结构。`memset()``memset()`
> 
> 为了解决这个问题，**Alpha 值在每像素的基础上显式设置为 255**，以确保背景不透明。此外，还对实际使用的函数进行了修改，以正确应用预期值。`clearDisplay()``memset()`
> 
> ### 💻 Updated Code
> void UVCPreview::clearDisplay() {
>     ENTER();
> 
>     ANativeWindow_Buffer buffer;
>     pthread_mutex_lock(&capture_mutex);
>     {
>         if (LIKELY(mCaptureWindow)) {
>                // code omitted
>         }
>     }
>     pthread_mutex_unlock(&capture_mutex);
> 
>     pthread_mutex_lock(&preview_mutex);
>     {
>        //////////////////////////////////////////////////////////////////////
>         if (LIKELY(mPreviewWindow)) {
>             if (LIKELY(ANativeWindow_lock(mPreviewWindow, &buffer, NULL) == 0)) {
>                 uint8_t *dest = (uint8_t *)buffer.bits;
>                 const int stride = buffer.stride * PREVIEW_PIXEL_BYTES;
>            
>                 for (int i = 0; i < buffer.height; i++) {
>                     uint8_t *row = dest;
>                     for (int j = 0; j < buffer.width; j++) {
>                         row[0] = 0;    // R (Red)
>                         row[1] = 0;    // G (Green)
>                         row[2] = 0;    // B (Blue)
>                         row[3] = 255;  // A (Alpha, fully opaque)
>                         row += 4; // Move to the next pixel
>                     }
>                     dest += stride;
>                 }
>           ////////////////////////////////////////////////////////////
>                 ANativeWindow_unlockAndPost(mPreviewWindow);
>             }
>         }
>     }
>     pthread_mutex_unlock(&preview_mutex);
> 
>     EXIT();
> }
> 多亏了关键字，我才能够找到解决问题的线索。我衷心感谢`memset` [@shengjunhu](https://github.com/shengjunhu) .

Some device need to align memory，It is recommended to use 'buffer.stride ' instead of 'buffer.width' .  
ANativeWindow_Buffer buffer;
if (0 == ANativeWindow_lock(mPreviewWindow, &buffer, nullptr)) {
    memset(buffer.bits, 0, buffer.stride * buffer.height * 4);
    ANativeWindow_unlockAndPost(mPreviewWindow);
}

shengjunhu, 2025-02-06:
> ## ✅ Solution
> When using `memset()` to set RGBA values, an issue was encountered where the **A (Alpha) value was not applied correctly.** This is suspected to be because `memset()` operates **on a byte-by-byte basis,** making it unable to properly set the RGBA (4-byte) structure.
> 
> To resolve this, the **Alpha value was explicitly set to 255 on a per-pixel basis** to ensure an opaque background. Additionally, modifications were made to the `clearDisplay()` function, where `memset()` is actually used, to correctly apply the intended values.
> 
> ### 💻 Updated Code
> void UVCPreview::clearDisplay() {
>     ENTER();
> 
>     ANativeWindow_Buffer buffer;
>     pthread_mutex_lock(&capture_mutex);
>     {
>         if (LIKELY(mCaptureWindow)) {
>                // code omitted
>         }
>     }
>     pthread_mutex_unlock(&capture_mutex);
> 
>     pthread_mutex_lock(&preview_mutex);
>     {
>        //////////////////////////////////////////////////////////////////////
>         if (LIKELY(mPreviewWindow)) {
>             if (LIKELY(ANativeWindow_lock(mPreviewWindow, &buffer, NULL) == 0)) {
>                 uint8_t *dest = (uint8_t *)buffer.bits;
>                 const int stride = buffer.stride * PREVIEW_PIXEL_BYTES;
>            
>                 for (int i = 0; i < buffer.height; i++) {
>                     uint8_t *row = dest;
>                     for (int j = 0; j < buffer.width; j++) {
>                         row[0] = 0;    // R (Red)
>                         row[1] = 0;    // G (Green)
>                         row[2] = 0;    // B (Blue)
>                         row[3] = 255;  // A (Alpha, fully opaque)
>                         row += 4; // Move to the next pixel
>                     }
>                     dest += stride;
>                 }
>           ////////////////////////////////////////////////////////////
>                 ANativeWindow_unlockAndPost(mPreviewWindow);
>             }
>         }
>     }
>     pthread_mutex_unlock(&preview_mutex);
> 
>     EXIT();
> }
> Thanks to the keyword `memset`, I was able to find a clue to solve the problem. I sincerely appreciate [@shengjunhu](https://github.com/shengjunhu) .

Some device need to align memory，It is recommended to use 'buffer.stride ' instead of 'buffer.width' .

ANativeWindow_Buffer buffer;
if (0 == ANativeWindow_lock(mPreviewWindow, &buffer, nullptr)) {
memset(buffer.bits, 0, buffer.stride * buffer.height * 4);
ANativeWindow_unlockAndPost(mPreviewWindow);
}